fabianschuiki/moore #108: A question about your project

giannissc, 2019-10-10:
Hi my name is John and I am also a PhD student looking to write a new HDL borrowing ideas from Rust and Clash and I was wondering about the nature of your project.

So if I am understanding it right moore is a reference compiler for both VHDL and Verilog leveraging LLHD.

But LLHD IR can be also leveraged by simulators to do simulations, run test, etc.

So is LLHD IR netlist-like format essentially that I could use for example as as an input for place and route algorithms to generate either FPGA bitstream or CAD models for fabrication?

fabianschuiki, 2019-10-12:
Hi John, glad to hear that you are looking into new HDLs, which is an exciting proposition! You are right about LLHD, it is an IR that captures the essentials of VHDL/Verilog and brings them into an LLVM-like SSA form. Moore is the reference compiler which takes SystemVerilog/VHDL (all highly experimental) and maps it to LLHD. The intention then is to take this LLHD and feed it into any of the commercial EDA tools. For example, a synthesizer would map this LLHD description to a list of connected logic gates of a specific manufacturing technology (a netlist). Right now none of these tools directly support LLHD. However they use "internal IRs" and my intention is to provide a mapping from LLHD to these IRs, allowing the tools to be fed LLHD via a lightweight translation step. In the hope that later on they will add native LLHD support. The output of these tools is a Verilog netlist, which is the de-facto standard of the industry.

fabianschuiki, 2019-10-12:
> Also you README file appears to be outdated. I have succesfully compiled system verilog code (an SR latch) using your website but havent been able to do so using moore itself.
> 
> The only command that i found relevant was `moore compile foo.sv` but I couldn't figure out how to get a foo.llhd file with that command. The -e flag is not recongnised. Could you help me with that as well?

Thanks for the hint. Indeed the version of moore that you can install via `cargo` is horribly outdated (v0.1). Since all is highly experimental, your best option is to clone this repository directly and build that. That's also what's online at llhd.io. I hope I can sort out the crates.io publishing soon.

fabianschuiki, 2019-10-12:
Closing for now. Feel free to reopen the issue!

closed 2019-10-12 by fabianschuiki

giannissc, 2019-10-12:
So for now LLHD is more of an IR converter than an IR optimizer/compiler.

For LLVM you would have something like this:

`Source Code -> Frontend Compiler -> LLVM IR -> LLVM Optimizer/Compiler -> Optimized LLVM IR -> Backend`

Whereas for LLHD it would be something like this:

`Source Code -> Frontend Compiler -> LLHD IR -> LLHD Converter -> EDIF/Verilog Netlist -> EDA tools`

giannissc, 2019-10-12:
I think that this is great solution as it allows you to be productive right away instead of waiting to develop all the components.

Would it make sense in the future to have something like this though:
`Source Code -> Frontend Compiler -> LLHD IR -> LLHD Optimizer/Compiler -> Optimized LLHD IR`

and then offer two different paths:
1. `Optimized LLHD IR -> IR Converter -> EDA tools`
2. `Optimized LLHD IR -> Floor Planning and Place & Route Backend -> GDS or STEP -> Format Converter`

Eventually all the heavy lifting when it comes to optimizations and checks could be carried out by LLHD (similar to how it is currently done with LLVM) and a collection of backends could be built for various cases e.g. FPGAs, PLDs, ASIC manufacturing.

This will allow for more higher level HDL to be built in the future by leveraging LLHD as their framework

giannissc, 2019-10-12:
As is the nature of my project I would have to follow the 2nd path and I would be interested in collaborating to take the project in both the EDA tools converter direction and the custom built Place & Route backends direction!

giannissc, 2019-10-12:
PS. I will download the git version and let you know if I have any more trouble :)

Also since i will be experimenting with the IR in the coming days if I happen to create good notes I would be happy to package them into a docs.md to go along with the repository

fabianschuiki, 2019-10-12:
I misunderstood. So the intention is to have a similar flow as you described for LLVM:
```
Source Code -> Frontend Compiler -> LLHD IR -> LLHD Optimizer/Compiler -> Optimized LLHD IR -> Synthesizer -> Gate-level Netlist (EDIF/Verilog) -> Placer/Router -> GDS 
```
In this case the backend is the usual EDA tool chain (synthesizer, placer/router). Ideally LLHD would be the "lingua franca" to take you from the high-level user-facing HDL through some common optimization to the flow adequate for your target (FPGA, PLD, ASIC).

The custom-built P&R direction is very interesting, but you also need synthesis capabilities that translate from LLHD into a collection of standard cells. There are some open source efforts to do this (although from Verilog, e.g. Yosys by Clifford Wolf), but it's a very challenging task. Synthesizers contribute an essential part to ASIC manufacturing in that they take several design space exploration decisions when going from LLHD/Verilog to a gate-level netlist (mainly architectures for arithmetic subcircuits, unreachable state elimination, etc.). While some of these could be promoted to a technology-agnostic opt pass in LLHD, most of these decisions are technology-dependent and will be hard to do in a works-for-everyone fashion. I'm very curious to see where your project takes you.

Also contributions in terms of notes/docs are very welcome, as they help others to get started with the project!

fabianschuiki, 2019-10-12:
By the way, check out the [Language Reference](http://llhd.io/spec.html) for details on the IR syntax and semantics. And I'd love to hear about any inconsistencies you might find!

giannissc, 2019-10-12:
Well in my case cause I won't be working with electronic circuits but rather fluidic ones and we are working with our own standard cells I was hoping I could use LLHD IR as a sort of a netlist format and see if could do synthesis/P&R in one pass.

`Source Code -> Frontend Compiler -> LLHD IR -> LLHD Optimizer/Compiler -> Optimized LLHD IR -> Floor Planning and Place & Route Backend -> GDS or STEP -> Format Converter`

I am still in the investigation phase (I have only started in September) so will how this pans out after studying the language reference.

Thank you for the quick reply btw!

giannissc, 2019-10-12:
[image]

This my most recent pipeline but some of the steps are wrong i believe