fabianschuiki/moore #126: sv-tests: Apply preprocessor macros to entire compilation unit

fabianschuiki, 2020-03-21:
Macros are currently limited to file scope. This is a sensible safety measure. However, moore should have an option to enable the default SystemVerilog behaviour of clobbering the entire compilation unit with macros. Or vice versa, an option to isolate each input file in its own compilation unit.

For example: https://symbiflow.github.io/sv-tests/#moore_parse|basejump|BaseJumpSTL_bsg_cache_to_dram_ctrl_rx

phsilva, 2020-07-21:
@fabianschuiki I was looking for low hanging fruits on `sv-tests` and found the many basejump STL tests failing and ended up discovering that it is because of the compilation unit treatment as describe here.

Any suggestion on the strategy to offer a solution to this? And let me know if this is worth doing right now.

fabianschuiki, 2020-07-22:
@phsilva It would be great to fix this! I've seen this compilation unit behaviour being used in various places, so I think it makes sense to get moore up to speed with this rather sooner than later. The fix could go something like this:

The main `moore` executable has the following loop to process the input files: https://github.com/fabianschuiki/moore/blob/9304e2c124590e7d0552144c26100235696a2e48/src/bin/moore.rs#L193-L196
To enable this compilation unit behaviour, we need a way to transfer the preprocessor state from the previous file to the next one, *if it is enabled*.

The preprocessor is defined here: https://github.com/fabianschuiki/moore/blob/9304e2c124590e7d0552144c26100235696a2e48/src/svlog/syntax/preproc.rs#L16-L42
From these fields, we want to transport the `macro_defs` and `dirs` from one preprocessor invocation to the next.

My suggestion would be the following:
- In `src/svlog/syntax/preproc.rs`:
    - Add a `State` struct which has fields for `macro_defs` and `dirs`
    - Add a `get_state` function to the preprocessor to get a copy of these fields wrapped in the `State` struct
    - Add a `Option<State>` parameter to `Preprocessor::new` to allow it to be optionally pre-loaded with a state
- In `src/bin/moore.rs`:
    - Add an option `--separate` that enables compiling each file as its own compilation unit
    - Pass the previous file's preprocessor state to the current preprocessor's `new()`, *if* the above flag is not set

phsilva, 2020-07-22:
@fabianschuiki awesome instructions, thanks. I will work on this and see where it goes. 👍