fabianschuiki/moore #142: Add arithmetic module

fabianschuiki, 2020-05-05:
Constant evaluation is pretty flimsy at the moment. The compiler essentially maps everything to a `BigInt` and then fiddles around with them, not performing any kinds of value truncation or similar. It would be beneficial to add a module/type that properly performs arithmetic with the SystemVerilog semantics. This could simply be `llhd::value::IntValue` for starters (would require a dependency update). At some point SV probably needs some custom ops, which we might either just `impl` onto `IntValue`, or just create our own version of the type.

programmerjake, 2020-05-05:
I wrote some code that could be adapted:
https://salsa.debian.org/Kazan-team/spectre-resistant-speculative-processor/-/blob/b894d88aea15f7dbe237a0aa25b0ca6345d95ca7/src/logic.rs#L391

you have my permission to use `logic.rs` under the Apache 2 OR MIT license even though it's currently LGPLv2.1+

programmerjake, 2020-05-05:
If you do use it, I would appreciate adding a link to the original source in a comment or something, along with the note about my giving you permission to relicense it

fabianschuiki, 2020-05-05:
Pretty neat, thanks! I'll take a look!