fabianschuiki/moore #168: Compile

tampler, 2020-05-21:
Hello Fabian

Started to learn about this project. This work is awesome! Thanks for contributing a lot of effort and building this!

I tried to compile an example from your paper with `moore` and got a bug. Just to let you know about this issue. Here's the [file](https://drive.google.com/open?id=1Iv2Jptlt0NJTm9-qGzJw2zH5nqzQ1pn)
```bash
bku@lap:~/work/hw/llhd/moore/0$ moore -O4 tb.sv -e acc_tb
compiler bug: lowering of call to expression not implemented
  --> tb.sv:13:3-8: 
   |   check(i, q);
```

fabianschuiki, 2020-05-22:
Hi @tampler, thanks for the heads up. There are a few parts of the examples in the paper that moore does not yet fully support, but was in the paper for the sake of completeness (and show how things would map to LLHD). I'm working on getting this done though :+1:. Changing this into a tracking issue for function calls.

tampler, 2020-05-22:
Sure take your time. I'll be keeping my eye on this project and your awesome work

fabianschuiki, 2020-05-22:
Thanks! Any SystemVerilog code snippets that you find which are rejected by moore are very welcome here on the issue tracker :smiley: !

commit cf33e0c9 (2022-02-07): codegen: Implement function calls

Add support for function calls, including the various types of passing
arguments in and out of functions.

Fixes #168.
closed 2022-02-08 by fabianschuiki