fabianschuiki/moore #170: Discussion - LLHD integration with other EDA

tampler, 2020-05-22:
Hello 

I'm interested in this project, since am working on a commercial version of a FIRRTL Simulator implemented in Scala for large SoC designs. Our goal is to support Rocket and Boom simulations in pure FIRRTL first, and then add SV. VHDL is not considered at all for a 2y roadmap.

We work to promote Chisel as the first class language for simulation and synthesis, which is kind of what you mentioned as a "future" in your paper.

We indeed developed our own IR, frontend parser, and a sim backend implementation, which is capable for massive parallelization and cluster simulation, kudos to JVM Ecosystem.

However, I see no value in building and supporting my own IR and would happily switch to LLHD or similar. I'm also looking at `LGraph` IR to be compatible with `Yosys` and `STA` tools.

Currently, we stuck to FIRRTL, since Berkeley added many useful passes on the FIRRTL level, including early area, timing and power estimation, DFT insertion, FPGA cluster simulation support, etc. More info about that [here](https://dl.acm.org/doi/pdf/10.5555/3199700.3199728) and [here](https://people.eecs.berkeley.edu/~biancolin/papers/dessert-fpl18.pdf). The later paper shows excellent results for integration with `Yosys` and combining both FIRRTL and Yosys passes to gain a higher level of design optimization.

So

1. I'm curious about your thoughts on integration LLHD with Chisel and LGraph compatible tools 
2. I'm also interested in a roadmap of `moore` and what are your plans and priorities on it?
3. Do you consider porting `moore` design to Scala ?

Sincerely,
Boris

fabianschuiki, 2020-05-22:
Hi @tampler 

> I'm interested in this project, since am working on a commercial version of a FIRRTL Simulator implemented in Scala for large SoC designs. Our goal is to support Rocket and Boom simulations in pure FIRRTL first, and then add SV. VHDL is not considered at all for a 2y roadmap.

That's an awesome endeavor!

> We work to promote Chisel as the first class language for simulation and synthesis, which is kind of what you mentioned as a "future" in your paper.
>
> We indeed developed our own IR, frontend parser, [...]

Indeed, one way of how LLHD can be useful is to have languages like Chisel/FIRRTL map to it rather than Verilog. Of course if you already use `LGraph` to push designs through other tools, then changing to LLHD might not be worthwhile just for the sake of changing.

> Currently, we stuck to FIRRTL, since Berkeley added many useful passes on the FIRRTL level, including early area, timing and power estimation, DFT insertion, FPGA cluster simulation support, etc. [...]

Yes this is indeed a very nice feat!

> 1. I'm curious about your thoughts on integration LLHD with Chisel and LGraph compatible tools

- LLHD is intended to make it easy to transport designs around. One thing on the todo list is conversion from/to FIRRTL, which would allow for the chain Chisel -> FIRRTL -> LLHD, but also SystemVerilog -> LLHD -> FIRRTL. (https://github.com/fabianschuiki/llhd/issues/124)
- Regarding Yosys specifically, conversion to/from RTLIL is on the todo list (https://github.com/fabianschuiki/llhd/issues/116). Ideally, Yosys will eventually gain a native LLHD frontend, which would allow it to ingest any other HDL that has LLHD support.
- Conversion from/to LGraph would be another step in this direction.

> 2. I'm also interested in a roadmap of `moore` and what are your plans and priorities on it?

The long-term plans are to make this a complete SV implementation (famous last words). In the short-term, I push towards getting the various [pulp-platform](https://github.com/pulp-platform) IPs to compile. At the same time, I'm trying to improve the [sv-tests](https://github.com/SymbiFlow/sv-tests) track record of moore.

> 3. Do you consider porting `moore` design to Scala ?

Not at all.

tampler, 2020-05-22:
Hi @fabianschuiki 
Thanks for detailed answers.

1. How far are you from delivering https://github.com/fabianschuiki/llhd/issues/124 ?

I may switch solely to LLHD in our simulation, if you can support FIRRTL input (any form) in LLHD
This would allow for us to simulate both Chisel and SV and a bit VHDL thru the LLHD layer, which will deliver a great benefit to entire ecosystem. I may also give you a free access to our simulator for research purposes to improve LLHD internals and SV mapping

2. Couldn't find your JIT simulator `llhd-blaze`. Is it a private/commercial or a not yet published project?

3. Yes, we are working to design chips in SV too. I'll try `moore` to compile our production designs and give you a feedback on that

Best,
Boris

fabianschuiki, 2020-05-22:
> @tampler
> 1. How far are you from delivering [fabianschuiki/llhd#124](https://github.com/fabianschuiki/llhd/issues/124) ?

The conversion tool (`llhd-conv`) and some early trials are set up, but no FIRRTL yet. I added the issue to the v0.14 milestone. That would make v0.14 an ecosystem integration release. You having a potential use for it is a good incentive to push for this rather sooner than later.

> I may switch solely to LLHD in our simulation, if you can support FIRRTL input (any form) in LLHD
> This would allow us to simulate both Chisel and SV and a bit VHDL thru the LLHD layer, which will deliver a great benefit to entire ecosystem. I may also give you a free access to our simulator for research purposes to improve LLHD internals and SV mapping

That'd be very nice!

> 2. Couldn't find your JIT simulator `llhd-blaze`. Is it a private/commercial or a not yet published project?

It's not yet published, but will be. It needs some clean-up and love before that, though.

> 3. Yes, we are working to design chips in SV too. I'll try `moore` to compile our production designs and give you a feedback on that

Thanks, that would be very appreciated.