fabianschuiki/moore #196: The output of moore from the example from llhd.io cannot be run on llhd-sim.

jaroslov, 2020-06-18:
I've installed from cargo the following binaries:
> ~/.cargo/bin/llhd-sim --version
> llhd-sim 0.3.1
> ~/.cargo/bin/moore --version
> moore 0.10.0
I then copy-and-pasted the example code from 'llhd.io' and synthesized it:
> moore accum.sv -e Accumulator > accum.llhd
When I try to simulate the resulting accum.llhd I get the following error:
> Parse error at line: 1, column: 84
> Unexpected `-`
> Expected `(`
[accum.zip](https://github.com/fabianschuiki/moore/files/4801613/accum.zip)

fabianschuiki, 2020-06-19:
`llhd-sim` currently uses an older version of `llhd` which is in need of updating to catch up with changes in recent versions (see https://github.com/fabianschuiki/llhd-sim/issues/8). This will be mostly a mechanical update, but requires some attention. Will be done soon. I think version 0.6.0 (e.g. a579ce7) of moore still used a compatible version of llhd. Thanks for the working example!

fabianschuiki, 2020-10-16:
This should now be working with moore v0.11 and llhd 0.14.1 :+1:

closed 2020-10-16 by fabianschuiki