fabianschuiki/moore #199: Compiler bug elaborating svlog with interface modports

dicta, 2020-07-16:
I've been reproducably able to generate compiler bugs on both v0.10.0 and current master (4a651ef) when elaborating systemverilog bus components that utilize interfaces somewhat heavily. The following snippet (and associated error message) is the most I've been able to reduce the test case so far:

```
interface bus
();
    logic clk;
    logic data;

    modport recv (
       input clk,
       input data
    );
endinterface

module testcase
#()
(
    bus.recv the_bus,
    output logic result
);

    always_ff @(posedge the_bus.clk)
        result <= the_bus.data;

endmodule

module top
();

    bus bus_inst();
    testcase test_inst ( .the_bus(bus_inst.recv) );

endmodule
```

When elaborating using `moore test.sv -e testcase`, the compiler generates sensible output. The error is caused by `moore input.sv -e top`, with the following result:

```
 ❯ target/debug/moore test.sv -e top

compiler bug: no ast node for n14 in the map
   = note: Needed to compute the type of port `the_bus`:
  --> test.sv:15:5-21:
   |
   |     bus.recv the_bus,
   |     ^^^^^^^^^^^^^^^^

You have encountered a compiler bug. Sorry about that! We would appreciate if you open an issue [1] and describe how you triggered the bug, together with a minimal snippet of code to reproduce it. Thanks!
[1]: https://github.com/fabianschuiki/moore

thread 'main' panicked at 'Other map contains: ModportName #14', <::std::macros::panic macros>:5:6
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Please note that I believe the issue to be caused by the following line:

` testcase test_inst ( .the_bus(bus_inst.recv) );`

Changing this line to the following (excluding the modport) results in reasonable output:

` testcase test_inst ( .the_bus(bus_inst) );`

What's the state of modport support in moore generally?

fabianschuiki, 2020-07-17:
Thanks for the excellent error report :+1:. The "no ast node" error hints at an internal bug caused by transitioning the compiler to a more robust handling of AST nodes. I will take a look.

> What's the state of modport support in moore generally?

Moore currently supports the use of modports to indicate the direction of interface signals. Code generation consults the selected modport of an interface to figure out what directions the signals have. Modport features beyond that are not yet implemented (but are definitely on the todo list :smiley:).

fabianschuiki, 2020-09-05:
Fixed in d7cf99b8.

closed 2020-09-05 by fabianschuiki