fabianschuiki/moore #203: Questions about moore-vhdl-syntax

m-kru, 2020-07-23:
Hi, I wanted to ask few questions about the moore-vhdl-syntax. I would really appreciate if you answer.

1. What is the current status?
2. How many of the constructs defined in the LRM is it able to parse? Rough percentage estimation would satisfy me.
3. Is it capable of parsing [UVVM](https://github.com/UVVM/UVVM), [OSVVM](https://github.com/OSVVM/OSVVM), [VUnit](https://github.com/VUnit/vunit), [PoC](https://github.com/VLSI-EDA/PoC)?
3. Is it capable of parsing files in parallel?
4. What is the performance? What is the average time needed to parse a file?
5. Why have you written your own parser? Why not use [rust_hdl](https://github.com/kraigher/rust_hdl)? Why divide a community that is already small?

fabianschuiki, 2020-07-24:
Hi @m-kru

> 1. What is the current status?

It does *something*, but it's by no means anywhere resembling completion. Since most of the ASIC/FPGA world seems to be pivoting towards SystemVerilog (and my daily work involves Verilog :wink:), I'm focusing on getting the SV part in shape for the time being. But VHDL is in the roadmap and will get a complete implementation one day.

> 2. How many of the constructs defined in the LRM is it able to parse? Rough percentage estimation would satisfy me.

I'd say about 60%.

> 3. Is it capable of parsing [UVVM](https://github.com/UVVM/UVVM), [OSVVM](https://github.com/OSVVM/OSVVM), [VUnit](https://github.com/VUnit/vunit), [PoC](https://github.com/VLSI-EDA/PoC)?

Cool pointers, thanks. Haven't tried to compile any of these yet.

> 4. Is it capable of parsing files in parallel?

Yes, you can use any of the parallelization techniques (e.g. the wonderful `rayon` crate) to speed things up.

> 5. What is the performance? What is the average time needed to parse a file?

It's pretty fast, but I don't have numbers. The time to parse is actually not that relevant in practice, since parsing is generally the trivial part of compilation, and far more time is spent in the downstream passes which try to make semantic sense of the HDL.

> 6. Why have you written your own parser? Why not use [rust_hdl](https://github.com/kraigher/rust_hdl)? Why divide a community that is already small?

`rust_hdl` was not around when I started work on the VHDL part of Moore.

m-kru, 2020-07-24:
> Since most of the ASIC/FPGA world seems to be pivoting towards SystemVerilog

It's hard to say how true this is. [VHDL vs Verilog](https://www.reddit.com/r/FPGA/comments/fu1cci/vhdl_vs_verilog/)

>  The time to parse is actually not that relevant in practice

I can't agree. Let me introduce a broader point of view. VHDL community is suffering because of numerous VHDL parser projects. There are some tries to tries to try to unite. To satisfy different usages the parser must be fast, for example in terms of LSP server parsing performance makes a difference.

Generally, it looks like whole LLHD project has huge potential. In my humble opinion you should mark `moore-vhdl-syntax` as legacy and in the future use rust_hdl. If you document how to fasten parsers with `moore-vhdl` (I guess this is the actual compiler processing the syntax tree) maybe someone would even does this.