fabianschuiki/moore #216: Failed installation

SJFeldt, 2020-11-11:
Hello,
I just tried to set up Moore on an Ubuntu 18.04 machine, using `cargo install moore` but unfortunate the building failed with the following error message:

```
...
   Compiling hibitset v0.6.3
   Compiling moore-common v0.11.0
   Compiling num-complex v0.3.1
   Compiling moore-vhdl-syntax v0.11.0
   Compiling moore-svlog-syntax v0.11.0
   Compiling moore-vhdl v0.11.0
   Compiling num v0.3.1
   Compiling llhd v0.14.3
   Compiling moore-svlog v0.11.0
error[E0277]: the trait bound `llhd::IntValue: std::convert::From<(usize, num::BigInt)>` is not satisfied
    --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/moore-svlog-0.11.0/codegen.rs:1111:39
     |
1111 |                 Ok(self.builder.ins().const_int((size, k.clone())))
     |                                       ^^^^^^^^^ the trait `std::convert::From<(usize, num::BigInt)>` is not implemented for `llhd::IntValue`
     |
     = help: the following implementations were found:
               <llhd::IntValue as std::convert::From<(usize, num_bigint::bigint::BigInt)>>
               <llhd::IntValue as std::convert::From<(usize, num_bigint::biguint::BigUint)>>
               <llhd::IntValue as std::convert::From<(usize, usize)>>
     = note: required because of the requirements on the impl of `std::convert::Into<llhd::IntValue>` for `(usize, num::BigInt)`

error[E0308]: mismatched types
    --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/moore-svlog-0.11.0/codegen.rs:1116:57
     |
1116 |                 .const_time(llhd::value::TimeValue::new(k.clone(), 0, 0))),
     |                                                         ^^^^^^^^^ expected struct `num_rational::Ratio`, found struct `num::rational::Ratio`
     |
     = note: expected struct `num_rational::Ratio<num_bigint::bigint::BigInt>`
                found struct `num::rational::Ratio<num::BigInt>`
     = note: perhaps two different versions of crate `num_rational` are being used?

error[E0277]: the trait bound `llhd::IntValue: std::convert::From<(usize, num::BigInt)>` is not satisfied
    --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/moore-svlog-0.11.0/codegen.rs:1495:62
     |
1495 | ...                   let lg2 = self.builder.ins().const_int((width, BigInt::from(lg2)));
     |                                                    ^^^^^^^^^ the trait `std::convert::From<(usize, num::BigInt)>` is not implemented for `llhd::IntValue`
     |
     = help: the following implementations were found:
               <llhd::IntValue as std::convert::From<(usize, num_bigint::bigint::BigInt)>>
               <llhd::IntValue as std::convert::From<(usize, num_bigint::biguint::BigUint)>>
               <llhd::IntValue as std::convert::From<(usize, usize)>>
     = note: required because of the requirements on the impl of `std::convert::Into<llhd::IntValue>` for `(usize, num::BigInt)`

error[E0277]: the trait bound `llhd::IntValue: std::convert::From<(usize, num::BigInt)>` is not satisfied
    --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/moore-svlog-0.11.0/codegen.rs:1501:65
     |
1501 | ...                   let lhs_ll = self.builder.ins().const_int((width, BigInt::one()));
     |                                                       ^^^^^^^^^ the trait `std::convert::From<(usize, num::BigInt)>` is not implemented for `llhd::IntValue`
     |
     = help: the following implementations were found:
               <llhd::IntValue as std::convert::From<(usize, num_bigint::bigint::BigInt)>>
               <llhd::IntValue as std::convert::From<(usize, num_bigint::biguint::BigUint)>>
               <llhd::IntValue as std::convert::From<(usize, usize)>>
     = note: required because of the requirements on the impl of `std::convert::Into<llhd::IntValue>` for `(usize, num::BigInt)`

error[E0277]: the trait bound `llhd::IntValue: std::convert::From<(usize, num::BigInt)>` is not satisfied
    --> ~/.cargo/registry/src/github.com-1ecc6299db9ec823/moore-svlog-0.11.0/codegen.rs:2328:48
     |
2328 | ...                   self.builder.ins().const_int((way_width, mask))
     |                                          ^^^^^^^^^ the trait `std::convert::From<(usize, num::BigInt)>` is not implemented for `llhd::IntValue`
     |
     = help: the following implementations were found:
               <llhd::IntValue as std::convert::From<(usize, num_bigint::bigint::BigInt)>>
               <llhd::IntValue as std::convert::From<(usize, num_bigint::biguint::BigUint)>>
               <llhd::IntValue as std::convert::From<(usize, usize)>>
     = note: required because of the requirements on the impl of `std::convert::Into<llhd::IntValue>` for `(usize, num::BigInt)`

error: aborting due to 5 previous errors

Some errors have detailed explanations: E0277, E0308.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `moore-svlog`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: failed to compile `moore v0.11.0`, intermediate artifacts can be found at `/tmp/cargo-installsWVyJI`

Caused by:
  build failed
```
Cargo Version: cargo 1.47.0 (f3c7e066a 2020-08-28)

Is it a problem with my system or an inconsistant code version?

fabianschuiki, 2020-11-11:
@SJFeldt thanks for reporting this! I can reproduce the issue on my machine as well. This is likely a breakage in the upstream `num-bigint` package. For the time being, tinstalling  the current master should work:
```
cargo install --git https://github.com/fabianschuiki/moore
```

SJFeldt, 2020-11-12:
Thanks, this seems to work!

closed 2020-11-12 by SJFeldt