fabianschuiki/moore #218: Compilation fails on DFF module

SeanMcLoughlin, 2020-12-31:
Source code:

```systemverilog
// File: dff.sv
module dff(clk, reset, d, q); 
    input clk;
    input reset;
    input d;
    output reg q;

    always_ff @(posedge clk or posedge reset) begin
        if (reset)
            q <= 1'b0;
        else
            q <= d;
    end 
endmodule
```

Command:
```
moore dff.sv -e dff
```

Error:
```
compiler bug: VarDeclName with weird parent PortDecl(PortDecl #25)
  --> dff.sv:2:8-11:
   | 
   |     input clk;
   |           ^^^ 
   = note: Encountered at src/svlog/typeck.rs:62
   = note: Needed to compute the type of the expression:
  --> dff.sv:7:22-25:
   | 
   |     always_ff @(posedge clk or posedge reset) begin
   |                         ^^^                        

You have encountered a compiler bug. Sorry about that! We would appreciate if you open an issue [1] and describe how you triggered the bug, together with a minimal snippet of code to reproduce it. Thanks!
[1]: https://github.com/fabianschuiki/moore
```

fabianschuiki, 2021-01-06:
This is an excellent catch, thanks!

The problem lies in this part of the `type_of` function: https://github.com/fabianschuiki/moore/blob/386fa012203a25e99290e5e0704785a423161422/src/svlog/typeck.rs#L56-L66

Moore is gradually moving towards doing more work on the AST rather than the HIR (which has turned out to be mostly just a copy of the AST with parent nodes resolved and `NodeId` scattered all over the place). This block is a remnant of that: type computation of most variable-declaration-like things has moved into this block. What's missing is an entry for `ast::AllNode::PortDecl` which should call the corresponding `type_of_int_port` that already exists. That'll likely require querying the port list of the enclosing module and looking up the `port_list::IntPort` that corresponds to this `PortDecl` (Complicated because SV has this ANSI/non-ANSI port list nonsense).

Would you feel comfortable tackling this issue?

SeanMcLoughlin, 2021-01-06:
Sure, I'll take a look at it.

PR #219 (2021-01-07): Fix bug with port decl outside of module params not parsing correctly
Fixes #218. I'm not sure if this is the correct fix, but I re-created the test in the `test` directory and it seems to parse unlike before.
closed 2021-01-07 by fabianschuiki