fabianschuiki/moore #228: systemverilog generate construct support

WonHoYoo, 2021-08-14:
Hi, I just tried moore today, and got the error on the below systemverilog module
``` systemverilog
//Function Unit
//Related to Integer Operation

module FU #(
  //generate adder number
  parameter add_num = 1,
  //generate multiplier number
  parameter mul_num = 1
  )(
  input clk,
  input rst_n,
  input [3:0] Op,
  input  [add_num-1:0][31:0] A_ADD,
  input  [add_num-1:0][31:0] B_ADD,
  output [add_num-1:0][31:0] C_ADD,

  input  [mul_num-1:0][31:0] A_MUL,
  input  [mul_num-1:0][31:0] B_MUL,
  output [mul_num-1:0][31:0] C_MUL
  );

  genvar idx;

  generate for (idx=0; idx < add_num; idx++) begin : gen_add
    vector_add vAdd
    (
      .clk,
      .rst_n,
      .A(A_ADD[idx]),
      .B(B_ADD[idx]),
      .C(C_ADD[idx])
    );
  end
  endgenerate

  generate for (idx=0; idx < mul_num; idx++) begin : gen_mul
    vector_mul vMul
    (
      .clk,
      .rst_n,
      .A(A_MUL[idx]),
      .B(B_MUL[idx]),
      .C(C_MUL[idx])
    );
  end
  endgenerate

endmodule
```
with below command
```bash
moore -e FU -o output.mlir -f mlir FU.sv vector_add.v vector_mul.sv
```
When printing the ast with the option, --dump-ast, it seems the moore compiler doesn't generate the module instantiation as much as the number of the for loop. 
Could you check on this? 
Thank you

fabianschuiki, 2021-08-16:
Thanks @WonHoYoo for reporting this! Do you by any chance have the output of Moore handy and can post it here for quick reference / sake of completeness?

WonHoYoo, 2021-08-18:
Hello, @fabianschuiki. 
Sorry for the late reply because of my heavy work schedule in my workplace.
Below is the output from the moore command. 

``` mlir
entity @FU (i1$ %clk, i1$ %rst_n, [1 x i32]$ %A_ADD, [1 x i32]$ %B_ADD, [1 x i32]$ %A_MUL, [1 x i32]$ %B_MUL) -> ([1 x i32]$ %C_ADD, [1 x i32]$ %C_MUL) {
    %0 = const i32 0
    %1 = [i32 %0]
    %2 = const time 0s
    drv [1 x i32]$ %C_ADD, %1, %2
    drv [1 x i32]$ %C_MUL, %1, %2
}
``` 
It seems the generate statement is ignored in the moore toolchain. Would there be anything I am missing on the moore options to apply for getting the generate statement work right? 
Thanks

fabianschuiki, 2021-08-18:
Hey @WonHoYoo, thanks for getting back to this with the output! This should definitely contain the `vAdd` and `vMul` instances as you expect, without the need to pass any special flags to Moore. Let me look into this bug.

fabianschuiki, 2021-08-18:
This is fixed in version 0.12.1 :+1:

commit f6169788 (2021-08-18): svlog: Fix generate regions being ignored

Fix an issue where generate regions in the Verilog code would be ignored
during emission, such that `generate for ... endgenerate` would not
produce any output, but a bare `for ...` would.

Fixes #228.
closed 2021-08-18 by fabianschuiki