fabianschuiki/moore #232: Add CIRCT HW/Comb/LLHD as backend

maerhart, 2021-09-09:
Add CIRCT as backend or replace the current Rust-LLHD backend with CIRCT. This allows us to leverage the whole CIRCT infrastructure within moore. There are, however, a few things the current backend supports that CIRCT does not.

* [ ] Link moore against CIRCTs C-API
* [ ] Generate rust bindings for the C-API using bindgen
* [ ] Write a wrapper to make creating operations, types and attributes easier and to hide unsafe code
* [ ] Add CIRCT assembly generation as in `codegen.rs`
* [ ] Possibly register and expose `llhd-sim` and other passes to the moore CLI

fabianschuiki, 2021-09-09:
Awesome! Check out the `mlir` branch, I've been starting some work there.

maerhart, 2021-09-09:
Wow what a quick response! Sorry, I haven't seen the branch before creating the issue, so I thought I'll just create a tracking issue to get an overview of what should be done. Looks really nice already!

fabianschuiki, 2021-11-21:
#232 implements a first version of this (the old LLHD crate stuff is still there, but we can strip that out soon!).

fabianschuiki, 2022-01-16:
Merged #234 which implements this.

closed 2022-01-16 by fabianschuiki