fabianschuiki/moore #233: Is there a plan to implent moore in c++?

hunterzju, 2021-11-12:
Excellent work! And I wonder if there is a plan to transform it to c++ implentation? Or how to do such a work?

fabianschuiki, 2021-11-12:
Thanks @hunterzju! LLHD is slowly but steadily moving into the [CIRCT project](https://github.com/llvm/circt), where it will enjoy an awesome community and be useful to a larger group of people. I am planning to migrate Moore over to use the LLHD implementation in CIRCT instead of the `llhd` crate, by linking directly to CIRCT (see the [mlir](https://github.com/fabianschuiki/moore/tree/mlir) branch which has some initial work). If that goes well, my ultimate long-term plan is to start moving the Moore-internal IRs (AST, RST, HIR, MIR) over to an MLIR-based implementation in CIRCT, possibly up to a complete replacement with MLIR/C++. There are a few things in how Moore/SV have to deal with nodes in the IR (everything is on-demand and cannot be pass-based) which require some attention on the MLIR side, but there's nothing preventing this.

Long story short, the ideal way forward would be:
- Link against CIRCT and directly emit LLHD using CIRCT
- Move codegen from Rust over to C++, which requires implementing the Moore MIR dialect in MLIR, and the SV type system
- Move type checking over to CIRCT, which requires implementing the Moore HIR dialect (or something similar) in MLIR
- Move name and syntactic ambiguity resolution over to CIRCT, which requires implementing the AST/RST in CIRCT
- Move parsing over to CIRCT (I would love to use a [Pika parser](https://arxiv.org/abs/2005.06444) for this, generated from a grammar file, with human guidance on how to resolve/represent ambiguities -- it makes no sense to handwrite all that SV because it becomes hard to adapt/troubleshoot, but human guidance is required because the grammar is so abysmally ambiguous)
- At this point Moore would be pretty much a pure MLIR implementation

hunterzju, 2021-11-13:
> Thanks @hunterzju! LLHD is slowly but steadily moving into the [CIRCT project](https://github.com/llvm/circt), where it will enjoy an awesome community and be useful to a larger group of people. I [...]

Thanks for your reply, I am a newbie in LLVM, but interesting in the work. I'll learn it and see if I can do something. Thanks again.

closed 2021-11-13 by hunterzju