fabianschuiki/moore #240: building error? cannot find MLIRExecutionEngine?

Joejiong, 2022-03-28:
-- Configuring done
CMake Error at llvm/install/lib/cmake/llvm/AddLLVM.cmake:515 (add_dependencies):
  The dependency target "MLIRExecutionEngine" of target
  "obj.CIRCTLLHDSimEngine" does not exist.
Call Stack (most recent call first):
  llvm/install/lib/cmake/mlir/AddMLIR.cmake:147 (llvm_add_library)
  cmake/modules/AddCIRCT.cmake:28 (add_mlir_library)
  lib/Dialect/LLHD/Simulator/CMakeLists.txt:28 (add_circt_library)

CMake Error at llvm/install/lib/cmake/llvm/AddLLVM.cmake:1818 (add_dependencies):
  The dependency target "mlir-cpu-runner" of target "check-circt" does not
  exist.
Call Stack (most recent call first):
  llvm/install/lib/cmake/llvm/AddLLVM.cmake:1839 (add_lit_target)
  test/CMakeLists.txt:40 (add_lit_testsuite)

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

shibizhao, 2022-03-29:
I have met the same problem~

maerhart, 2022-03-31:
With commit 9bd6d54fcfc8914cc2e225af4a91146ff13a9646 Moore changed the upstream CIRCT repo. If you cloned this repo before this commit and then updated/pulled it, did you call something like
```
$  git submodule sync
$  git submodule update --init --recursive --remote
```
to update the submodule URL and commit of CIRCT? Alternatively, doing a fresh git clone should work. Otherwise, please let me know.

shibizhao, 2022-04-01:
@maerhart  Thanks for your reply. I make a fresh git clone, however, it still reports the same bug like this:

Make Error at llvm/install/lib/cmake/llvm/AddLLVM.cmake:515 (add_dependencies):
  The dependency target "MLIRExecutionEngine" of target
  "obj.CIRCTLLHDSimEngine" does not exist.
Call Stack (most recent call first):
  llvm/install/lib/cmake/mlir/AddMLIR.cmake:147 (llvm_add_library)
  cmake/modules/AddCIRCT.cmake:28 (add_mlir_library)
  lib/Dialect/LLHD/Simulator/CMakeLists.txt:28 (add_circt_library)

CMake Error at llvm/install/lib/cmake/llvm/AddLLVM.cmake:1820 (add_dependencies):
  The dependency target "mlir-cpu-runner" of target "check-circt" does not
  exist.
Call Stack (most recent call first):
  llvm/install/lib/cmake/llvm/AddLLVM.cmake:1841 (add_lit_target)
  test/CMakeLists.txt:41 (add_lit_testsuite)

shibizhao, 2022-04-03:
Thanks for your advice! 
I add the `"host"` in the `cmake` command of llvm-build, and it works well during the `cmake` command of circt-build. However, when I execute the command:
`cmake --build . --target install`

It reports some bugs:
```
[ 66%] Built target obj.CIRCTStaticLogicOps
[ 66%] Built target CIRCTStaticLogicOps
[ 67%] Building CXX object lib/Conversion/AffineToStaticLogic/CMakeFiles/obj.CIRCTAffineToStaticLogic.dir/AffineToStaticLogic.cpp.
o
In file included from /data/shibizhao/docker/eda/bizhao/moore/circt/lib/Conversion/AffineToStaticLogic/AffineToStaticLogic.cpp:31:
0:
...
/home/shibizhao/data/docker/eda/bizhao/moore/circt/llvm/mlir/include/mlir/Transforms/DialectConversion.h:720:37: error: variable ‘
callback’ has function type
         [=](Operation *op) { return callback(cast<OpT>(op)); });
                                     ^~~~~~~~
/home/shibizhao/data/docker/eda/bizhao/moore/circt/llvm/mlir/include/mlir/Transforms/DialectConversion.h:720:37: error: variable ‘
callback’ has function type
```
just like this.
Looking forward to your reply!

Joejiong, 2022-04-03:
> It reports some bugs:

I'm actually clone the circt separately, but It seems search the standard dialect header somewhere.

thx

Joejiong, 2022-04-03:
Solved!!!

Finally solved it, it is because not install with correct path. 

Sorry, new to rust.