fabianschuiki/moore #244: Build and Install problems in moore

DeepFlyingSky, 2022-04-20:
Follow the [README.md](https://github.com/fabianschuiki/moore/blob/master/README.md) to build the lllvm and then circt.

It will raise an error
```
CMake Error at llvm/install/lib/cmake/llvm/AddLLVM.cmake:1820 (add_dependencies):
  The dependency target "mlir-cpu-runner" of target "check-circt" does not
  exist.
Call Stack (most recent call first):
  llvm/install/lib/cmake/llvm/AddLLVM.cmake:1841 (add_lit_target)
  test/CMakeLists.txt:41 (add_lit_testsuite)

```

It's because the llvm upstream no longer support auto build mlir-cpu-runner, unless you check the LLVM_NATIVE_ARCH

[Do not build llc and mlir-cpu-runner to run MLIR tests](https://reviews.llvm.org/rG46b1e837169885bda02792682020d65d8ca21818)

We can change -DLLVM_TARGETS_TO_BUILD=""  to -DLLVM_TARGETS_TO_BUILD="host" in cmake settings.

And  I still cannot run `cargo install moore` success, in spite of `cargo build` get approved.
I'm new to rust and learning it now. The error message is follow

```
  cargo:rustc-link-lib=dylib=pthread
  cargo:rustc-link-lib=dylib=m
  cargo:rustc-link-lib=dylib=z
  cargo:rustc-link-lib=dylib=tinfo
  cargo:rerun-if-changed=wrapper.h

  --- stderr
  wrapper.h:20:10: fatal error: 'mlir-c/Dialect/Standard.h' file not found
  wrapper.h:20:10: fatal error: 'mlir-c/Dialect/Standard.h' file not found, err: true
  thread 'main' panicked at 'Unable to generate bindings: ()',
```

gussmith23, 2022-07-12:
Same issue!

DeepFlyingSky, 2022-07-13:
OS message is follow
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal
```

I try the new cmake build setting you just edit,
add with `-DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++ -DLLVM_ENABLE_LLD=ON `
to enable compile fast.

samerogers, 2022-08-23:
I am having the same issue as the one mentioned above. Even after rolling back both my LLVM and CIRCT builds and using crate install --path . I am unable to build Moore. There appear to be some significant updates on the LLVM side that have broken the Moore build. Is there a timeline for a Moore update?

Phantom1003, 2022-10-30:
I have the same problem. Is there any update?