fabianschuiki/moore #245: Failing to build on Mac

gussmith23, 2022-07-12:
Some of the issues I'm running into:
1. The clang version on Mac is too old, and can't compile some of the code that `build.rs` attempts to compile. Fixed by setting `CXX=` and `CC=` to newer clang binaries.
2. The system libs hard-coded in `build.rs` are not the right ones for Mac -- presumably they're for Linux? I just hard-coded the Mac ones, but a better solution would probably be to call `llvm-config` in `build.rs`?
3. Finally, I'm not at all sure what's happening here:

```
  = note: "cc" "-m64" "-arch" "x86_64" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.0.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.1.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.10.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.11.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.12.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.13.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.14.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.15.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.2.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.3.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.4.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.5.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.6.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.7.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.8.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.moore.c16926be-cgu.9.rcgu.o" "/Users/gus/moore/target/debug/deps/moore-dfd3fb1c7ea007a6.79hhl54c4dou9ls.rcgu.o" "-L" "/Users/gus/moore/target/debug/deps" "-L" "/Users/gus/circt/build/lib" "-L" "/Users/gus/circt/llvm/build/lib" "-L" "/Users/gus/moore/target/debug/build/moore-circt-sys-306822630f73fa78/out" "-L" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib" "/Users/gus/moore/target/debug/deps/libpretty_env_logger-dee5a9513857e167.rlib" "/Users/gus/moore/target/debug/deps/libenv_logger-cd997496271be5d5.rlib" "/Users/gus/moore/target/debug/deps/libtermcolor-15024d6cc5a290ce.rlib" "/Users/gus/moore/target/debug/deps/libhumantime-5758ed6bc3244712.rlib" "/Users/gus/moore/target/debug/deps/libquick_error-525ce0b56cf0634c.rlib" "/Users/gus/moore/target/debug/deps/libmoore-96769668c50af5a7.rlib" "/Users/gus/moore/target/debug/deps/libmoore_vhdl-f23c4a3bcf8d73ef.rlib" "/Users/gus/moore/target/debug/deps/libmoore_vhdl_syntax-376d7b0ff2910e41.rlib" "/Users/gus/moore/target/debug/deps/liblazy_static-d09aab148284a89a.rlib" "/Users/gus/moore/target/debug/deps/libmoore_svlog-de898f383bd26331.rlib" "/Users/gus/moore/target/debug/deps/libitertools-a1c1382831559089.rlib" "/Users/gus/moore/target/debug/deps/libbit_vec-9a5c52ebf761c059.rlib" "/Users/gus/moore/target/debug/deps/libmoore_svlog_syntax-10f3152e9f3063d9.rlib" "/Users/gus/moore/target/debug/deps/libbacktrace-286a53071bf71392.rlib" "/Users/gus/moore/target/debug/deps/libobject-c55da26aa2fec167.rlib" "/Users/gus/moore/target/debug/deps/libaddr2line-40bf4763fba34b71.rlib" "/Users/gus/moore/target/debug/deps/libgimli-8ef9252a3739f15e.rlib" "/Users/gus/moore/target/debug/deps/librustc_demangle-60a820a61e61f68c.rlib" "/Users/gus/moore/target/debug/deps/libmoore_common-97a14e3dd3baa20b.rlib" "/Users/gus/moore/target/debug/deps/libmemmap2-d6d954653b682c93.rlib" "/Users/gus/moore/target/debug/deps/libtyped_arena-287fb1dfe435b06e.rlib" "/Users/gus/moore/target/debug/deps/libmoore_circt-7a83f3945fe27e1d.rlib" "/Users/gus/moore/target/debug/deps/libmoore_circt_sys-41e0f70c0c4ad7a1.rlib" "/Users/gus/moore/target/debug/deps/libllhd-0ee803c56db46ad0.rlib" "/Users/gus/moore/target/debug/deps/libitertools-f455c17a3c869424.rlib" "/Users/gus/moore/target/debug/deps/libhibitset-2804345f6742d6de.rlib" "/Users/gus/moore/target/debug/deps/librayon-12019c651ebfa618.rlib" "/Users/gus/moore/target/debug/deps/librayon_core-bbf5d9b9419bfd08.rlib" "/Users/gus/moore/target/debug/deps/libnum_cpus-a1f407655e8bef36.rlib" "/Users/gus/moore/target/debug/deps/libcrossbeam_deque-160d7de5e21ff7a4.rlib" "/Users/gus/moore/target/debug/deps/libcrossbeam_epoch-ae7f33ca87eb230b.rlib" "/Users/gus/moore/target/debug/deps/libmemoffset-281457c0a90ecc07.rlib" "/Users/gus/moore/target/debug/deps/libscopeguard-ac4032aa7f5beabf.rlib" "/Users/gus/moore/target/debug/deps/libcrossbeam_channel-e7b8628c1b5a8cae.rlib" "/Users/gus/moore/target/debug/deps/libcrossbeam_utils-c10a5883b7754b3d.rlib" "/Users/gus/moore/target/debug/deps/libonce_cell-1c1b8933057e04d5.rlib" "/Users/gus/moore/target/debug/deps/libeither-3fcfd5c5e136f285.rlib" "/Users/gus/moore/target/debug/deps/libatom-d049c3a028a9f33f.rlib" "/Users/gus/moore/target/debug/deps/libnum-128808663017f564.rlib" "/Users/gus/moore/target/debug/deps/libnum_iter-ee441ad5d3da2429.rlib" "/Users/gus/moore/target/debug/deps/libnum_rational-8d7a349e389dfc65.rlib" "/Users/gus/moore/target/debug/deps/libnum_complex-d7bf4e0791e25ed9.rlib" "/Users/gus/moore/target/debug/deps/libnum_bigint-66c61ba017e6210d.rlib" "/Users/gus/moore/target/debug/deps/libnum_integer-1de31340ddc025d6.rlib" "/Users/gus/moore/target/debug/deps/libnum_traits-0c72a9a4ab04a453.rlib" "/Users/gus/moore/target/debug/deps/libregex-d83ae8a6abf330d0.rlib" "/Users/gus/moore/target/debug/deps/libaho_corasick-0f0bc9d5c60441f5.rlib" "/Users/gus/moore/target/debug/deps/libmemchr-ec581d75a5a916fe.rlib" "/Users/gus/moore/target/debug/deps/libregex_syntax-9e6c6888eea69ccd.rlib" "/Users/gus/moore/target/debug/deps/liblalrpop_util-665fb784106b3051.rlib" "/Users/gus/moore/target/debug/deps/libserde-0c16dc2a9e2d03bd.rlib" "/Users/gus/moore/target/debug/deps/libclap-16cf5a379d0d3edd.rlib" "/Users/gus/moore/target/debug/deps/libvec_map-6b5d9cec2c1d08ae.rlib" "/Users/gus/moore/target/debug/deps/libtextwrap-f673a3ab9aa9b40c.rlib" "/Users/gus/moore/target/debug/deps/libunicode_width-1bdeec44e91f1830.rlib" "/Users/gus/moore/target/debug/deps/libstrsim-941a4300d1e02281.rlib" "/Users/gus/moore/target/debug/deps/libbitflags-13d2a907925fc12c.rlib" "/Users/gus/moore/target/debug/deps/libatty-8dbc8e0b8961267e.rlib" "/Users/gus/moore/target/debug/deps/liblibc-68425e5c6092e35c.rlib" "/Users/gus/moore/target/debug/deps/libansi_term-e62f0d7f8b039f96.rlib" "/Users/gus/moore/target/debug/deps/liblog-0693ee8797391f31.rlib" "/Users/gus/moore/target/debug/deps/libcfg_if-57b943692f42b17b.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libstd-056ec5f8da85a5c6.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libpanic_unwind-0eca8ccbb05d41e8.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libobject-f3df534d6cb2a8a3.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libmemchr-7508553746157663.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libaddr2line-41b3db98dcc27d51.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libgimli-bbd7b7eecf756628.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libstd_detect-b5576df482f2a799.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/librustc_demangle-02015773483edf96.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libhashbrown-dda3dbde44c7a624.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/librustc_std_workspace_alloc-7573f724261122d0.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libunwind-b003af69f65f627c.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/libcfg_if-02094c1160da984c.rlib" "/Users/gus/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/lib/liblibc-57149dc33cede4d9.rlib" "/Users/gus/.r [... cut ...] faultlibs"
  = note: Undefined symbols for architecture x86_64:
            "llvm::itaniumDemangle(char const*, char*, unsigned long*, int*)", referenced from:
                llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) in libmoore_circt_sys-41e0f70c0c4ad7a1.rlib(Signals.cpp.o)
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Any thoughts? Totally perplexed by the reference to Itanium.

gussmith23, 2022-07-12:
alright, i was able to (hackily) get it working with these changes:

```diff
diff --git a/src/circt-sys/build.rs b/src/circt-sys/build.rs
index b50db1c..ed3002f 100644
--- a/src/circt-sys/build.rs
+++ b/src/circt-sys/build.rs
@@ -26,7 +26,7 @@ fn main() {
         .unwrap_or_else(|_| llvm_dir.join("build"));
 
     // CIRCT/LLVM/MLIR libraries
-    let lib_dirs = [circt_build_dir.join("lib"), llvm_build_dir.join("lib")];
+    let lib_dirs = [circt_build_dir.join("lib"), llvm_build_dir.join("lib"), PathBuf::from("/usr/local/Cellar/llvm/13.0.1_1/lib/")];
     let lib_names = [
         "CIRCTCAPIComb",
         "CIRCTCAPIHW",
@@ -84,8 +84,9 @@ fn main() {
 
     // System libraries
     let system_libraries = [
-        "stdc++", // llvm-config --system-libs --link-static
-        "rt", "dl", "pthread", "m", "z", "tinfo",
+"LLVM-13",
+"m", "z", "curses", "xml2"
+
     ];
     for name in &system_libraries {
         println!("cargo:rustc-link-lib=dylib={}", name);
```

fabianschuiki, 2022-07-12:
That's fantastic thanks. I haven't tried the build on OSX in a while. Probably worth adding a CI job for it. Out of curiosity: the manual homebrew cellar path you had to add -- would that show up in `llvm-config --system-libs --link-static`? That might be a portable way of getting at the necessary system libs.

gussmith23, 2022-07-12:
Yup, `llvm-config` provides that info on Mac. Here are various calls to `llvm-config` and their results on Mac:

```sh
➜  moore git:(master) ✗ /usr/local/Cellar/llvm/13.0.1_1/bin/llvm-config --libs --link-shared       
-lLLVM-13
➜  moore git:(master) ✗ /usr/local/Cellar/llvm/13.0.1_1/bin/llvm-config --system-libs --link-shared

➜  moore git:(master) ✗ /usr/local/Cellar/llvm/13.0.1_1/bin/llvm-config --libs --link-static
-lLLVMWindowsManifest -lLLVMXRay -lLLVMLibDriver -lLLVMDlltoolDriver -lLLVMCoverage -lLLVMLineEditor -lLLVMXCoreDisassembler -lLLVMXCoreCodeGen -lLLVMXCoreDesc -lLLVMXCoreInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMX86Desc -lLLVMX86Info -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyUtils -lLLVMWebAssemblyInfo -lLLVMSystemZDisassembler -lLLVMSystemZAsmParser -lLLVMSystemZCodeGen -lLLVMSystemZDesc -lLLVMSystemZInfo -lLLVMSparcDisassembler -lLLVMSparcAsmParser -lLLVMSparcCodeGen -lLLVMSparcDesc -lLLVMSparcInfo -lLLVMRISCVDisassembler -lLLVMRISCVAsmParser -lLLVMRISCVCodeGen -lLLVMRISCVDesc -lLLVMRISCVInfo -lLLVMPowerPCDisassembler -lLLVMPowerPCAsmParser -lLLVMPowerPCCodeGen -lLLVMPowerPCDesc -lLLVMPowerPCInfo -lLLVMNVPTXCodeGen -lLLVMNVPTXDesc -lLLVMNVPTXInfo -lLLVMMSP430Disassembler -lLLVMMSP430AsmParser -lLLVMMSP430CodeGen -lLLVMMSP430Desc -lLLVMMSP430Info -lLLVMMipsDisassembler -lLLVMMipsAsmParser -lLLVMMipsCodeGen -lLLVMMipsDesc -lLLVMMipsInfo -lLLVMLanaiDisassembler -lLLVMLanaiCodeGen -lLLVMLanaiAsmParser -lLLVMLanaiDesc -lLLVMLanaiInfo -lLLVMHexagonDisassembler -lLLVMHexagonCodeGen -lLLVMHexagonAsmParser -lLLVMHexagonDesc -lLLVMHexagonInfo -lLLVMBPFDisassembler -lLLVMBPFAsmParser -lLLVMBPFCodeGen -lLLVMBPFDesc -lLLVMBPFInfo -lLLVMAVRDisassembler -lLLVMAVRAsmParser -lLLVMAVRCodeGen -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMARMDisassembler -lLLVMARMAsmParser -lLLVMARMCodeGen -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMAMDGPUDisassembler -lLLVMAMDGPUAsmParser -lLLVMAMDGPUCodeGen -lLLVMAMDGPUDesc -lLLVMAMDGPUUtils -lLLVMAMDGPUInfo -lLLVMAArch64Disassembler -lLLVMAArch64AsmParser -lLLVMAArch64CodeGen -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMOrcJIT -lLLVMMCJIT -lLLVMJITLink -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMOrcTargetProcess -lLLVMOrcShared -lLLVMDWP -lLLVMSymbolize -lLLVMDebugInfoPDB -lLLVMDebugInfoGSYM -lLLVMOption -lLLVMObjectYAML -lLLVMMCA -lLLVMMCDisassembler -lLLVMLTO -lLLVMCFGuard -lLLVMFrontendOpenACC -lLLVMExtensions -lPolly -lPollyISL -lLLVMPasses -lLLVMObjCARCOpts -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMFrontendOpenMP -lLLVMDWARFLinker -lLLVMGlobalISel -lLLVMMIRParser -lLLVMAsmPrinter -lLLVMDebugInfoMSF -lLLVMDebugInfoDWARF -lLLVMSelectionDAG -lLLVMCodeGen -lLLVMIRReader -lLLVMAsmParser -lLLVMInterfaceStub -lLLVMFileCheck -lLLVMFuzzMutate -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMBinaryFormat -lLLVMTableGen -lLLVMSupport -lLLVMDemangle
➜  moore git:(master) ✗ /usr/local/Cellar/llvm/13.0.1_1/bin/llvm-config --system-libs --link-static
-lm -lz -lcurses -lxml2
➜  moore git:(master) ✗ /usr/local/Cellar/llvm/13.0.1_1/bin/llvm-config --libdir     
/usr/local/Cellar/llvm/13.0.1_1/lib
```