fabianschuiki/moore #250: Build fails due to missing `Standard.h` file in mlir-c/Dialect

vani2620, 2023-02-22:
Trying to build on Windows 10 version 21H2 build 19044.2604.

LLVM and CIRCT build and compile properly, but Moore HDL seems to not build properly. [Here](https://pastebin.com/cymFh215)'s the command line output.

Portion of interest appears to be
`  --- stderr
  wrapper.h:20:10: fatal error: 'mlir-c/Dialect/Standard.h' file not found
  wrapper.h:20:10: fatal error: 'mlir-c/Dialect/Standard.h' file not found, err: true
  thread 'main' panicked at 'Unable to generate bindings: ()', C:\Users\vani2\.cargo\registry\src\github.com-1ecc6299db9ec823\moore-circt-sys-0.14.0\build.rs:113:10
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to compile `moore v0.14.0`, intermediate artifacts can be found at `C:\Users\vani2\AppData\Local\Temp\cargo-installXkIIcC``

vani2620, 2023-04-12:
Any ideas as to what's causing this?

Eymay, 2023-04-24:
Try `cargo install --path .` as discussed in #248