fabianschuiki/moore #254: State and move to CIRCT

ecstrema, 2025-01-21:
I just read the LLHD/Moore paper. Incredible job, very well done. 

Noe it seems that this is slowly moving towards MLIR/CIRCT. 

So am I understanding well that the Moore compiler is slowly rewritten in C++ to emit MLIR instead, and that LLHD is being replaced by CIRCT?

If yes, should that be added to the Readme? And maybe this repository archived?

I am having a hard time understanding what's happening with this whole ecosystem lately. It seems that a lot of new promising tools emerge, and then merge into CIRCT et al. without stating anywhere that the original project has moved and/or is not maintained anymore. 

Also, from my understanding from the paper, Moore is a compiler targeting LLHD. Now I'm reading in the forums that it is an MLIR dialect? Or are these 2 different things with the same name?

Thank you

fabianschuiki, 2025-01-27:
Hey @ecstrema, thank you very much!

You're exactly right: the Moore compiler is being split up and reorganized as part of the CIRCT project, based on MLIR. LLHD has been integrated into CIRCT by @maerhart quite a while ago. More recently, we have started to integrate the Slang SystemVerilog frontend in CIRCT, which provides parsing and elaboration for the language. The IR/internals of the Moore compiler are being moved into CIRCT as the Moore MLIR dialect, which is generated from the AST the Slang produces, and acts as an intermediate dialect along the way to the core CIRCT dialects.

So in essence, the Moore compiler consisted of a preprocessor, parser, type checker, elaboration, and lowering to LLHD. Most of this is now handled by Slang, and part of the elaboration and lowering to LLHD are now part of CIRCT as the "Moore" dialect.

ecstrema, 2025-01-27:
Thank you. I will then close this issue. 

I would love to see all this info in the readme. It is very valuable for someone getting into the community. 

When I have time, I'll try to make a diagram of the different projects, with their states.

closed 2025-01-27 by ecstrema